ntu-aris/MMAUD #2: Can you provide the relative translation and rotation relationship of each sensor during the collection of the dataset?

lcjBegin, 2024-03-30:
The lack of relative positional relationships between various sensors makes it difficult to coordinate the application of multimodal information: translation and rotation relationships between livox-avia and Leica Nova MS60 (ground truth), lidar and Leica Nova MS60 (ground truth), fisheye cameras (left and right), and Leica Nova MS60 (ground truth)

snakehaihai, 2024-05-08:
MMAUD V1 (collected at rooftop with total raw data of 250G, after compression is < 80G) is already open. 
V2 (Collect at carpark for CVPR challenge trial, roughly 800G) is compressing
V3 (Collect at carpark for CVPR challenge Final, roughly 500G) is pending for compressing

For V1 you may try to download.

This is an issue that we know. Thats why we collected V2 and V3 sequences with MS60 calibrated to the Rig. 

For the calibration, there is inter-camera calibration https://drive.google.com/drive/folders/1wk-c5xVX6701WNI_In1ba3_D4LSjRYv5. 

Initially, I thought this work was about using NN to fit the multi-sensor observations to the ground truth. Since you asked about it, I'll have to think of a solution for this. Between each lidar, there is minimal FOV overlap. The CAD file is already available in the readme link for you to download for raw calibration. For refinement, I'll try running MLCC from the HKU team and see if I can get a good result. I'll update this after ICRA.

closed 2024-10-30 by snakehaihai

Byte247, 2024-12-11:
Is there any update on the extrinsic calibration between the used sensors?

snakehaihai, 2024-12-11:
> Is there any update on the extrinsic calibration between the used sensors?

Hi. For now, I think you may stick to CAD design. The MLCC result is far from the actual hardware design from our inhouse tests.