ntu-aris/MMAUD #5: .zip not downloading

efectual, 2024-06-23:
**Describe the BUG**
.bag's available but .zips not available
**To Reproduce**
macos safari
https://ntu-aris.github.io/MMAUD/
click .zip
nothing

**Site link or Repository**

snakehaihai, 2024-06-26:
ok uploading

snakehaihai, 2024-06-27:
All Zip should be ready to download if you can not download it. do let me know

closed 2024-06-27 by snakehaihai

molchanov-vs, 2024-09-07:
[image]
All .zip links don't work. Please check them

snakehaihai, 2024-09-08:
@molchanov-vs

Hi. For V2 and V3, it is not in the ICRA paper contribution but is a separate CVPR challenge.

You may look at this google drive to download the section.  https://drive.google.com/drive/folders/1wk-c5xVX6701WNI_In1ba3_D4LSjRYv5

Rgs

Shenghai Yuan

Nasakenaii, 2025-02-12:
Hello, 
The google drive link doesn't work anymore for the hard and moderate, could you please provide another link ? 

Thank you.

snakehaihai, 2025-02-12:
> Hello, The google drive link doesn't work anymore for the hard and moderate, could you please provide another link ?
> 
> Thank you.

Hi. I just tested this URL and it still works. Could it be that you are in China and didn't use VPN?