ntu-aris/MMAUD #8: Open Source on the 2D Object Detection and 3D pos estimations.

Besherz, 2024-10-21:
Dear Manager,

It's a great job! 
Could you upload the **code** of the _2D object detection_ part and _3D position estimations_? 
Based on the datasets, I could not achieve the best result of the Yolov5 yet. Sincerely hope to get the origin code. 
It could help me a lot. Thanks!

snakehaihai, 2024-10-21:
Hi Besherz

We'll upload the annotated YOLO bounding boxes and meta parameters shortly. The model used is standard, and you can use our annotations for retraining.

Rgs

Shenghai Yuan

yizhuoyang, 2024-10-23:
Hi Besherz
Thank you for your interest in our work. You can access the data for 2D object detection via this link: https://drive.google.com/drive/folders/1_LpPyIfETQS-k2vlSsbzI9pzyVzZScSx?usp=sharing
We follow the official code of yolov5 to train the model: https://github.com/ultralytics/yolov5
You can test it with the uploaded data, and feel free to reach out if you run into any issues while implementing the model.

Besherz, 2024-10-23:
@snakehaihai @yizhuoyang Thanks! There are lots of job you've done! I'll try to digest all lately. Thanks very much.

closed 2024-10-23 by Besherz