ntu-aris/MMAUD #11: track ground

ChancellorNG, 2024-12-13:


snakehaihai, 2024-12-14:
Our next goal is to work on a multi-drone scenario, planned for March 2025. Currently, we are targeting the operation of three drones. And for now, we are trying to clear out the legal paperwork. 

For multi-UAV tracking, you'll need multiple Leica MS60 trackers, which can be complex to set up. However, Leica offers a certified accuracy of 2mm, which is highly reliable. In practice, it is always within 1 cm accuracy. We use it a lot in SLAM GT.

Another option is using RTK GPS. While RTK GPS provides up to 1cm accuracy in ideal conditions, its accuracy can drop to 2-10cm in less optimal scenarios. If the drones are flown near tall buildings, the performance can degrade even further.

A third option is similar to our multi-campus dataset published in CVPR 2024. With each drone carrys a LIDAR and measure against a known map. and based on LIDAR matching result, then we generate trajectory. The expected accuracy should be able to reach somewhere around 2-5cm.

ChancellorNG, 2024-12-14:
Thank you for your patient explanation，another phenomenon that we are interested in is that in the real UAV perception environment, the performance of the voice array does not seem to be so reliable. What do you think of this phenomenon? Is the sound array necessary in the mode used?

snakehaihai, 2024-12-16:
It depends on your target application. On the battlefield, the first indication of a UAV's presence is often the humming sound it produces, even before it becomes visible to camera or LIDAR. If it is a sensitive subject, you may contact me by shyuan@ntu.edu.sg for further discussion

ChancellorNG, 2024-12-16:
Got it, we will be happy to share with you the interesting phenomena we observe later

closed 2024-12-16 by snakehaihai