ntu-aris/MMAUD #12: The parameters of inter-camera calibration

ZHCH111, 2025-01-02:
MMAUD V1 (collected at rooftop with total raw data of 250G, after compression is < 80G) is already open. 
V2 (Collect at carpark for CVPR challenge trial, roughly 800G) is compressing
V3 (Collect at carpark for CVPR challenge Final, roughly 500G) is pending for compressing

For V1 you may try to download.

This is an issue that we know. Thats why we collected V2 and V3 sequences with MS60 calibrated to the Rig. 

For the calibration, there is inter-camera calibration https://drive.google.com/drive/folders/1wk-c5xVX6701WNI_In1ba3_D4LSjRYv5. 

Initially, I thought this work was about using NN to fit the multi-sensor observations to the ground truth. Since you asked about it, I'll have to think of a solution for this. Between each lidar, there is minimal FOV overlap. The CAD file is already available in the readme link for you to download for raw calibration. For refinement, I'll try running MLCC from the HKU team and see if I can get a good result. I'll update this after ICRA.

_Originally posted by @snakehaihai in https://github.com/ntu-aris/MMAUD/issues/2#issuecomment-2100401124_

ZHCH111, 2025-01-02:
您好，请问一下Calibration results 
====================
Camera-system parameters:
cam0 (/left/image):
type: <class 'aslam_cv.libaslam_cv_python.DistortedOmniCameraGeometry'>
distortion: [-0.35828593 0.34304531 -0.00007972 -0.00060307] +- [0.06862911 0.34622041 0.00162955 0.00182931]
projection: [ 2.94142475 1844.98646292 1845.58033306 615.02487057 507.23461275] +- [0.01935148 0.3460135 
0.3458821 0.70074792 0.59580864]
reprojection error: [0.000001, 0.000005] +- [0.072684, 0.071740]
中distortion和projection中各个参数分别代表什么

snakehaihai, 2025-01-03:
> # 您好，请问一下Calibration results
> Camera-system parameters: cam0 (/left/image): type: <class 'aslam_cv.libaslam_cv_python.DistortedOmniCameraGeometry'> distortion: [-0.35828593 0.34304531 -0.00007972 [...]

参考这个

https://github.com/ethz-asl/kalibr

https://github.com/ethz-asl/kalibr/blob/master/aslam_cv/aslam_cameras/test/OmniCameraGeometry.cpp

distortion
 

0.35828593  k1（径向畸变系数）描述光学系统的径向畸变，导致图像边缘区域发生拉伸或收缩
,
0.34304531  k2（径向畸变系数）描述光学系统的径向畸变，导致图像边缘区域发生拉伸或收缩
,

0.00007972  p1（切向畸变系数） 描述相机镜头与成像平面不完美对齐造成的切向畸变。
,
0.00060307  p2（切向畸变系数） 描述相机镜头与成像平面不完美对齐造成的切向畸变。

 
projection

2.94142475  (xi,相机模型参数  ，通常描述相机光心到图像平面之间的几何关系）
,
1844.98646292   焦距 (focal length)，表示相机成像平面在x方向上的缩放比例。
,
1845.58033306  焦距 (focal length)，表示相机成像平面在y方向上的缩放比例。
,
615.02487057   x主点坐标 (principal point)，表示成像平面上的光心位置
,
507.23461275   t主点坐标 (principal point)，表示成像平面上的光心位置

closed 2025-01-03 by snakehaihai