ntu-aris/MMAUD #14: Request for Calibration Data in MMAUD Dataset

minaunal, 2025-03-19:
Dear team,

First of all, thank you for making the MMAUD dataset publicly available. It is a valuable resource for research in UAV detection and tracking.

I noticed that the dataset does not include calibration data, which is essential for certain tasks such as accurate localization and trajectory estimation. Would it be possible to provide the camera and sensor calibration files? Having access to these would significantly enhance the usability of the dataset for many researchers.

We truly appreciate your efforts and contributions to the community. Looking forward to your response.

Best regards,
Mina Unal

snakehaihai, 2025-03-19:
For calibration data, please refer to the google drive fisheye_calibration.zip  https://drive.google.com/drive/folders/1wk-c5xVX6701WNI_In1ba3_D4LSjRYv5

closed 2025-03-19 by snakehaihai

snakehaihai, 2025-03-20:
Hi For camera intrinsic and extrinsic they are all the same. I did not move in between camera for different sequence