ntu-aris/MMAUD #15: MMAUD Rooftop Simple V1 doesn't work with the provided calibration data

ilydglb, 2025-03-19:
Calibration data provided works correctly with the dataset in the Google Drive. But we need the calibration data for V1 Rooftop Simple datasets which are different drone model datasets. Are they the same? We wonder because same code including the calibration data does not work with V1 Rooftop Simple datasets. Thank you so much for your interest.

closed 2025-03-19 by ilydglb

snakehaihai, 2025-03-20:
Hi for the camera calibration they are the same. The only difference is where GT leica is positioned.  Do share with me your issues. 

Rgs
Shenghai Yuan

snakehaihai, 2025-03-26:
Hi 
If you are referring to in between sensor calibration data. here is the info

[image]

[image]