ntu-aris/MMAUD #16: Question about the 'Detection_new' folder in the MMAUD dataset

1TXR, 2025-03-19:
Hello, I'm currently working on reproducing the AV-FDTI project, but I've encountered an issue regarding the dataset. The project configuration files reference a 'Detection_new' folder, and after carefully reading the code, I understand that this folder should contain annotation information about whether drones exist in each image (1 for presence, 0 for absence).

I've downloaded and extracted the Mavic3.zip file, but I couldn't find this 'Detection_new' folder. I'm wondering if this folder exists in the MMAUD dataset? If yes, how can I obtain it? If not, how does the AV-FDTI project handle this part of the data?

Thank you for your help!

yizhuoyang, 2025-03-20:
Hi
The Detection_new dataset is not included in MMAUD. In AV-FDTI, the branch responsible for predicting the presence of drones in each image follows a self-supervised approach. This annotation is generated using a pretrained YOLO detection network, similar to the one mentioned in [this issue](https://github.com/ntu-aris/MMAUD/issues/8). If YOLO detects a drone, the label is set to 1; otherwise, it is set to 0.

snakehaihai, 2025-03-20:
I`ll invite yizhuo to help answer the issues.

closed 2025-03-20 by snakehaihai