ntu-aris/MMAUD #29: 没找到audio data

helloday0115, 2025-07-01:
I downloaded the UG2+ challenge and V1 series datasets, but found that there was no audio data in either, even neither in the readme file. Could you please tell me where the original audio data is?

snakehaihai, 2025-07-01:
Hi,

For the MMAUD V1 dataset, the audio data should be available within the ROS bag itself. Please check the rostopics.

[image]

Regarding the UG2+ challenge data (MMAUD V2 and V3): the audio signals were too weak to be of practical use, so we did not open-source the full ROS bag. Instead, we only released the segmented multi-modal data.

If you still wish to access the original ROS bag for  MMAUD V2 and V3, feel free to send me a link where I can upload it.  However, please be aware that the raw bag is quite large and can be extremely challenging to work with.

For hint on how to use visual and audio data, please refer to here https://github.com/yizhuoyang/AV-FDTI/?tab=readme-ov-file

Best regards,
Shenghai Yuan

helloday0115, 2025-07-02:
Yes, I would like the original 4 channel audio data. Could you please share with me a downloadable link? It could be on platforms like Baidu Cloud Drive or Google Drive, etc.

 

 
 
image.png (view on web)

snakehaihai, 2025-07-02:
> Yes, I would like the original 4 channel audio data. Could you please share with me a downloadable link? It could be on platforms like Baidu Cloud Drive or Google Drive, etc.
> […](#)

[image]

Press this. For V1 its already there

For V2 and V3, we did not upload the raw bag data and do not have it stored online, as these were not part of the original ICRA contribution (ICRA only cover V1). We are also not covering online storage costs for V2 and V3.
If you still need the data, feel free to send me a Baiduyun or other upload link, and I’ll upload it there for you.

helloday0115, 2025-07-02:
Thanks, that would be great.😀
Could you please upload the audio data of V2 and V3 to this link? 

https://c.wss.pet/s/he2w43istgl 

I would be really grateful.

  

  
image.png (view on web)

snakehaihai, 2025-07-02:
> Thanks, that would be great.😀
> Could you please upload the audio data of V2 and V3 to this link?&nbsp;
> 
> 
> https://c.wss.pet/s/he2w43istgl&nbsp;
> 
> 
> I would be really grateful.
> […](#)

[image]

No place to deposit

helloday0115, 2025-07-02:
How about this link?
 
https://c.wss.pet/s/he336ciqihx
 
It looks like it can work for data collection.😊
 
(To be honest, I’ve only used links for sharing data before, so this is my first time trying one for collection.)

  

 
https://c.wss.pet/s/he2w43istgl 
 
  
image.png (view on web)