ntu-aris/MMAUD #32: There is no visible audio dataset in the file system format downloads

DanielCohenn, 2025-07-14:
I downloaded multiple V1 datasets and the audio was not present in the file system.

[image]

If the audio is only available in the .ros file format please add this to the README. Also, I would be happy if you could add a quick guide on how to extract the datasets using the ROSBag SW and how to save it to numpy format.

snakehaihai, 2025-07-14:
Hi Daniel

Please refer to this post for solutions https://github.com/ntu-aris/MMAUD/issues/30

For audio, it's a bit tricky because there's no standard way to split it, and different methods can lead to different results. That's why we provide the original audio source so others can try it themselves. For other common sensors, we do save the processed data.

If you have other issues, do leave a message here. I`ll try my best to answer it.

Rgs
Shenghai Yuan

closed 2025-07-14 by snakehaihai

snakehaihai, 2025-07-14:
@DanielCohenn  We will try to find our sample audio split and GT with the audio. Will update you know in 24 hours.

DanielCohenn, 2025-07-14:
@snakehaihai Thank you very much for all the work ! 
I will be waiting for an update.

DanielCohenn, 2025-07-15:
@snakehaihai 
Thank you very much, I will look at this shortly.